mesh-iit/ipts #13: Validate mc4plus c-shape support in IPTS

davidetome, 2023-02-06:
## Task description
I need to test the support added in IPTS  testing the new 10` mc4plus c-shape` coming from the supplier.: 

- https://github.com/icub-tech-iit/purchases/issues/2043#issuecomment-1418886581

cc @pattacini @maggia80 @Fabrizio69 

## Definition of Done
- IPTS SW validated
- boards tested
- new release of IPTS produced

davidetome, 2023-04-07:
Ciao @Fabrizio69 ,

 are the new mc4plus c-shape arrived? Just to now, and plan this activity w/ @Nicogene 👍🏻 

cc @pattacini @maggia80

davidetome, 2023-04-07:
/remind me to plan this issue on Apr 16

Fabrizio69, 2023-04-07:
> Ciao @Fabrizio69 ,
> 
> 
> 
>  are the new mc4plus c-shape arrived? Just to now, and plan this activity w/ @Nicogene 👍🏻 
> 
> 
> 
> cc @pattacini @maggia80 

Hi @davidetome , I asked yesterday for the delivery date (see [purchase issue](https://github.com/icub-tech-iit/purchases/issues/2043#issuecomment-1499215217))

davidetome, 2023-05-17:
Today I tested 3 new boards w/ success, see 

- https://github.com/icub-tech-iit/proto/issues/2251#issuecomment-1551358334

Tomorrow I'll finalize the validation and I'll generate a new release of IPTS

cc @Nicogene @pattacini

davidetome, 2023-05-18:
Done w/ this PR : 

- https://github.com/icub-tech-iit/ipts/pull/19

Going to produce the new releease

closed 2023-05-18 by davidetome