mesh-iit/ipts #24: AMC-BLDC - add test support

davidetome, 2023-09-21:
As per the title, we need to add test support for `AMC-BLDC` boards.

DoD

- [ ] test firmware released
- [ ] new IPTS release produced

cc @Nicogene @maggia80 @marcoaccame

fgarini, 2024-07-23:
[image]
[image]
Kit in foto inscatolato e messo nell'armadio DEV in corridoio

MSECode, 2025-03-06:
Closing this EPIC since the test fw and new ipts version `v1.5.2`  has been released.
Check this PRs:
- test-firmware: https://github.com/robotology/icub-firmware/pull/570
- ipts release: https://github.com/icub-tech-iit/ipts/releases/tag/v1.5.2

closed 2025-03-06 by MSECode