mesh-iit/ipts #25: FAP - add test support

davidetome, 2023-09-21:
As per the title, we need to add test support for `FAP` boards.

DoD

- [ ] test firmware released
- [ ] new IPTS release produced

cc @Nicogene @maggia80 @marcoaccame

davidetome, 2023-09-26:
adding test support

- https://github.com/icub-tech-iit/ipts/compare/master...davidetome:ipts:fix/add-fap

to be checked an PR opened

davidetome, 2023-09-29:
PR opened, newIPTS release wi;; be produced as soon as it's merged

- https://github.com/icub-tech-iit/ipts/pull/26

cc @pattacini @Nicogene

davidetome, 2023-09-29:
The new IPTS release which adds FAP5 test support has been produced : 

- https://github.com/icub-tech-iit/ipts/releases/tag/v1.5.0

cc @pattacini @Nicogene @maggia80 @Fabrizio69

closed 2023-09-29 by davidetome

pattacini, 2023-09-29:
Sw released at:
- https://github.com/icub-tech-iit/ipts/releases/tag/v1.5.0