mesh-iit/ipts #33: AMC-BLDC - add tests using temporary test jig

davidetome, 2024-01-09:
## Task description
Following #31 , we need to add the tests that can be implemented using the debug setup provided by @fgarini (the final test jig will not be available soon).

Tests to be implemented : 

- Test Hall sensors
- Test Encoder
- Test Fault
- Test I2C

## Definition of Done

- [ ] Tests implemented in the test FW
- [ ] PR opened vs `robotology/icub-firmware`

cc @Nicogene @pattacini  @fgarini

davidetome, 2024-01-16:
adding tests , WIP : 

- https://github.com/davidetome/icub-firmware/commit/6ebf12c525f57143dc5125b04bc6e1db871e974f

cc @Nicogene

davidetome, 2024-01-18:
Done w/ this PR : 

- https://github.com/robotology/icub-firmware/pull/460

cc @Nicogene @pattacini @marcoaccame

fgarini, 2024-01-26:
great @davidetome! Let me give you a quick update and a suggestion:
In order to test the PWM outputs, while we wait for the official AMCxJIG do you think it may help if we use the part of the AUX-AMC-TEST (just arrived) that implements a similar burden (47R instead of 22R, with series inductance)?
I can provide support on this.