mesh-iit/ipts #34: AMC-BLDC - add  test support w/ temporary test jig

davidetome, 2024-01-18:
## Task description
Following #33 we need to start adding test support in IPTS starting from the temporary test jig.

## Definition of Done
- [ ] Temporary test support added and tested

cc @Nicogene  @pattacini

davidetome, 2024-01-19:
Started adding test support : 

- https://github.com/icub-tech-iit/ipts/compare/master...fix/%2334

cc @Nicogene

davidetome, 2024-01-24:
Added test support, I will test it tomorrow w/ the setup : 

- https://github.com/icub-tech-iit/ipts/commits/fix/%2334/

cc @Nicogene @fgarini @maggia80

davidetome, 2024-01-26:
Support added :and tested :  

- https://github.com/icub-tech-iit/ipts/commit/7be2d4c03a3269a8bb0e94fe660edc742fba61df

[image]

cc @Nicogene @maggia80

davidetome, 2024-01-26:
Done w/ this PR : 

- https://github.com/icub-tech-iit/ipts/pull/35

cc @Nicogene @pattacini

closed 2024-01-26 by davidetome

fgarini, 2024-01-26:
great @davidetome! Let me give you a quick update and a suggestion:
In order to test the PWM outputs, while we wait for the official AMCxJIG do you think it may help if we use the part of the AUX-AMC-TEST (just arrived) that implements a similar burden (47R instead of 22R, with series inductance)?
I can provide support on this.