mesh-iit/ipts #36: AMC - prepare test firmware project

davidetome, 2024-02-06:
## Task description
As per the title, prepare the `embot` based test project in Keil (`icub-firmware`) w/ a very basic first test on the CAN bus.

## Definition of Done
- [x] project created
- [x] project tested
- [x] PR opened in `icub-firmware devel`

cc @valegagge @marcoaccame @maggia80

MSECode, 2025-02-11:
I've started to create the project under `icub-firmware\emBODY\eBcode\arch-arm\board\amc\test\proj` for the tests on the `amc` board

MSECode, 2025-02-13:
Starting from a template for the `amc`  I've checked that the project is working fine as shown by the screen below. Thus now we have just to add the test for the CAN communication as requested by the tasks in the DoD.

[image]

MSECode, 2025-02-26:
Continuing the work shown in the previous comment we've continued to update the test project for the amc board and we have finalized the first test, i.e. the test for the CAN communication, which is actually working as expected as shown by the image below:

[image]

Moreover, starting from the template project of the amc board, we have decided to define the test class as a singleton and keep on that all the functions to be used then in the main file for executing the tests.

MSECode, 2025-02-27:
Pushed the fw here: https://github.com/MSECode/icub-firmware/tree/feature/amctests

closed 2025-03-05 by MSECode