mesh-iit/ipts #38: AMC-BLDC - finalize the test support with the official test jig

davidetome, 2024-02-06:
## Task description
Following #34, we need to finalize the test support with the official test jig

## Definition of Done
- [ ]  Add PWM output test to [test FW](emBODY/eBcode/arch-arm/board/amcbldc/test)
- [ ] Test the whole testplan using IPTS 
- [ ] A new release of IPTS has been produced

cc @valegagge @marcoaccame @maggia80

davidetome, 2024-02-06:
This morning I spoke w/ @maurizbo and he forecasted the jig available not before the 15th of March

cc @valegagge

davidetome, 2024-02-06:
/remind me on the 15th of March

fgarini, 2024-07-23:
Jig wired and ready to be used! see https://github.com/icub-tech-iit/ipts/issues/24#issuecomment-2245263310

fgarini, 2024-08-08:
I will add in this issue a description of the features of the JIG for the AMC-BLDC together with a description of the tests with the limit requirements for the pass/fail.
cc @MSECode

MSECode, 2025-01-23:
Regarding the test of the PWM for the amc-bldc test procedure we were able to add this test and we have checked that we are actually activating the PWM phases as shown by the image below where a current is shown on the oscilloscope:

[image]

For now we just send 50% of the max pwm on all the 3 phases.

In order to finalize the test we need to add the check of the currents on the adc.

MSECode, 2025-03-05:
Release created: https://github.com/icub-tech-iit/ipts/releases/tag/v1.5.2

Closing sice DoD satisfied.

closed 2025-03-06 by MSECode