mesh-iit/ipts #39: AMC-BLDC - option bytes

davidetome, 2024-02-06:
## Task description
Check the possibility of setting the option bytes automatically at the end of the test.

## Definition of Done
- [ ] feature added and tested

cc @valegagge @marcoaccame @maggia80

marcoaccame, 2024-02-12:
Hi @davidetome, the [`embot::hw::bsp::specialize()`](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/board/amcbldc/bsp/embot_hw_bsp_amcbldc.cpp#L920-L928) of the `amcbldc` calls a [verification](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/board/amcbldc/bsp/embot_hw_bsp_amcbldc.cpp#L891-L918) of the FLASH banks that uses a reading of the option bytes with `HAL_FLASHEx_OBGetConfig()`. 

I believe that there is a function also to set them.

davidetome, 2024-02-12:
> Hi @davidetome, the [`embot::hw::bsp::specialize()`](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/board/amcbldc/bsp/embot_hw_bsp_amcbldc.cpp#L920-L928) of the [...]

Thank you very much @marcoaccame !! I will take into account this as soon as I will work on it! 👍🏼

davidetome, 2024-06-18:
I found some references also here : 

- https://community.st.com/t5/stm32-mcus/how-to-program-stm32-option-bytes-with-the-hal-api/ta-p/49660

I added a function in the test FW to test it, but I'm facing issues w/ the debugger (flash timeout)

@marcoaccame / @fgarini are these the correct option bytes to be set?

[image]

cc @valegagge

fgarini, 2024-06-18:
Ciao @davidetome a suggestion may be to check if any additional precaution is needed for the H7 compared to the example you follow.
Here's the specific reference manual for the H7:
https://www.st.com/resource/en/reference_manual/rm0399-stm32h745755-and-stm32h747757-advanced-armbased-32bit-mcus-stmicroelectronics.pdf
In particular the section for the option bytes management is the 4.4

marcoaccame, 2024-06-18:
ciao @davidetome,  the above picture does not seem to me of the `amcbldc` but rather of a dual core stm32h7xx of the `amc` / `amcfoc` board.

for the `amcblcd` the option bit we need for sure is the one that configures FLASH as a single bank. I don't know the name of the option bit on cubeprogrammer, only the name in the code that checks it:

```C++
void verify_flash_bank()
{
    FLASH_OBProgramInitTypeDef odb = {0};   
    HAL_FLASHEx_OBGetConfig(&odb);
    
    uint8_t detectedbanks = ((odb.USERConfig & FLASH_OPTR_DBANK) == FLASH_OPTR_DBANK) ? 2 : 1;
```

fgarini, 2024-06-18:
[image]
this should be unchecked for having a single bank as for the indication above

marcoaccame, 2024-06-18:
> I found some references also here to do this using HAL API :
> 
> * https://community.st.com/t5/stm32-mcus/how-to-program-stm32-option-bytes-with-the-hal-api/ta-p/49660

this is interesting. just an idea for the future: to improve the above code so that it checks the dual back option bit and if not what is wanted it corrects it.

davidetome, 2024-06-18:
> ciao @davidetome, the above picture does not seem to me of the amcbldc but rather of a dual core stm32h7xx of the amc / amcfoc board.

ciao @marcoaccame , it's my fault I uploaded the incorrect screenshot...tomorrow I'll check 👍🏼

davidetome, 2024-06-21:
I added a function in the test FW to set needed option bytes : 

- https://github.com/davidetome/icub-firmware/commit/e43e16dc697f3e9c57f1a94120a9ec42765abe5d

```cpp
	HAL_StatusTypeDef testOB(){
			uint8_t data[8] {0};		
			uint8_t res {0};
			data[0] = 0xAA;

			FLASH_OBProgramInitTypeDef OB;

			HAL_FLASH_Unlock();
		  HAL_FLASH_OB_Unlock();

			OB.OptionType = OPTIONBYTE_USER;
		  OB.USERType = OB_USER_DBANK | OB_USER_BOR_LEV;
			OB.USERConfig = OB_DBANK_128_BITS | OB_BOR_LEVEL_4;

			if ( HAL_FLASHEx_OBProgram(&OB) != HAL_OK )
		  {
			  HAL_FLASH_OB_Lock();
			  HAL_FLASH_Lock();
				data[0] = 0xBB;
				sendCan(data);
			  return HAL_ERROR;
		  }

			sendCan(data);

		  HAL_FLASH_OB_Lock();
		  HAL_FLASH_Lock();
			return HAL_OK;
		}
```

Anyway, I didn't test it since I faced problems with the [test FW](https://github.com/robotology/icub-firmware/pull/460) which do not communicate on the CAN bus anymore.

It sounds very strange since I [tested the FW](https://github.com/icub-tech-iit/ipts/issues/34#issuecomment-1911667488) on the temporary test jig as shown in the [PI17 review demo](https://github.com/icub-tech-iit/fix/issues/1486#issuecomment-1910437503).

cc @valegagge @MSECode @marcoaccame @fgarini @maggia80

MSECode, 2024-08-29:
With @marcoaccame and @fgarini we actually check which option byte should be enabled and how to set them. However, the fw that tests it need to be re-checked. In the meanwhile I'll move back this issue to triage and when we are going to work on this I'll re-set it in progress.

MSECode, 2025-01-15:
I've update the code for the `amcbldc` tests with this commit: https://github.com/MSECode/icub-firmware/commit/5ed60895597433def3d23d20bbaf95485d99b746

Now the tests on the option bytes is working and it has been done accordingly to what discussed w/ @fgarini  in here: https://github.com/icub-tech-iit/dev/issues/762#issuecomment-2277498840

Moreover, related to the updates in that commit I have some doubt regarding these lines: https://github.com/MSECode/icub-firmware/blob/5ed60895597433def3d23d20bbaf95485d99b746/emBODY/eBcode/arch-arm/board/amcbldc/test/src/embot_app_skeleton_os_test.cpp#L376-L442 where we test the HALL sensors and the ABZ encoder. 
I thought to modify the previous code with the methods made available by the lower level code by Zini included in `encoder.c` and `pwm.c`, which is then called by the `embot` library. Thus now, my doubt is this one: I run the tests and they all pass as before. Therefore, I'm wondering if my updates are correct or if the code was ok as it was before. I've done those changes since looking at the structure of the code and seeing that the embot library and the lower level code were making available those methods, it was cleaner to just call those methods avoiding code duplications.
I would ask to @marcoaccame if we can revise that one of these days.

MSECode, 2025-01-17:
Here a test report showing that now all tests that are currently available are passing fine.

[0035_14962.D.txt](https://github.com/user-attachments/files/18457386/0035_14962.D.txt)

In order to finalize the activity I'd like to add here the minute of the discussion I've planner w/ @marcoaccame. Once my doubts are solved I'll add a comment and close the issue. 
Anyway, the test for the option bytes is now working with no issues.

valegagge, 2025-01-17:
Hi @MSECode ,
did you verify with the programmer as you suggested? thanks!

MSECode, 2025-01-27:
> Hi [@MSECode](https://github.com/MSECode) , did you verify with the programmer as you suggested? thanks!

Yes, I verified it with the debugger and everything seems to work fine.
I'm going then to push the latest changes and open the PRs.

closed 2025-01-30 by valegagge