mesh-iit/ipts #40: BLDC motor cod.7117 modifica indicazione procedure di test

Miche19, 2024-04-30:
Bisognerebbe sostituire alcune indicazioni:

**1)**

quelle attuali

[image]

con queste

[image]

**2)**

quelle attuali
[image]

con queste 
[image]

Speed positivo gira in senso antiorario / Speed negativo gira in senso orario

**3)**
 Se riteniamo sempre necessario individuare il cod. IIT per l albero campione da usare come verifica in fase di test.
Al momento nell`attrezzatura non è presente

[image]

cc: @Fabrizio69 @davidetome @Elio75 @fbiggi

Miche19, 2024-05-02:
@fbiggi mi ha fornito questi codici per l`alberino campione da poter utilizzare in fase di test 
ECUB_003_P_012-ECUB_003_P_009-ICUB3_001_P_220 
Uno di questi può andare bene

Guardando le giacenze delle **ultime** revisioni a gestionale purtroppo non ne abbiamo nessuno

- 
[image]

- 
[image]

- 
[image]

A questo punto @Elio75 ti chiederei di aprire una commessa di officina per la realizzazione di un pezzo di uno di questi codici (valuta te quello che preferisci)

cc: @Fabrizio69

Elio75, 2024-05-03:
Tutte 3 alberi hanno tolleranza g6. Per questo caso ho fatto la commessa [MMWS 2024-029- ID8746 ](https://github.com/icub-tech-iit/proto/issues/2840#issue-2277698657) contenete ECUB_003_P_009 cod.15768 rev A1 mi sembra più semplice come lavorazione.

CC @Miche19 @Fabrizio69 @fbiggi

Miche19, 2024-05-08:
Bisognerebbe inserire, forse come prima cosa nella procedura di collaudo, la verifica che sia inciso il codice fornitore e S/N sullo statore e rotore

[image]
[image]

cc: @davidetome @Fabrizio69

Miche19, 2024-05-27:
@Elio75 ha fatto produrre da [MWS](https://github.com/icub-tech-iit/proto/issues/2840) l alberino campione da utilizzare per il collaudo dei motori
**cod.IIT 15768.A1** 
Scaricato a gestionale questo componente e aggiunto all`attrezzatura di collaudo

cc: @davidetome @Fabrizio69

AntonioAzocar, 2024-09-11:
La stessa procedura di collaudo è utilizzata anche per i motori cod. **7116**,  quindi le modifiche segnalate da @Miche19 , sarebbero  da applicare per entrambe le procedure.