mesh-iit/ipts #46: Check the 2foc procedure

valegagge, 2024-07-29:
The supplier www.mwfep.com complaints regarding the USB CAN driver.

We provided the mini USB CAN device that needs a driver that is no longer available on the ESD site.
Verify if the procedure provides the correct driver. If not we need to update the kit `cod.4569.E - 2FOC_ADC, IIT - Testing equipment for 2FOC boards` with the newest CAN device.

**Dod**
The procedure is correct

MSECode, 2024-09-27:
### Here the update

Yesterday with @fgarini and @AntonioAzocar we run the whole procedure for testing the 2foc boards using the equipment given back by mwfep with code `2FOC cod. 4569` and here I'll report some notes regarding what done:

- first of some of the procedure steps, even if they are all correct in what they do, should be revised
- when using the goldenboard it is impossible to run the whole procedure with the same board and this is a problem since in the second phase the test bootloader and fw are not flashed at the beginning of the procedure, so that should be added as mentioned at the point above
- the board without the TH components in the test equipment should be changed or labeled with a SN
- all the tests of the procedure done with the goldenboard have been passed successfully, thus we sent an email to `mwfep` to ask them for checking the top and bottom of the 2foc boards they produced since they might be forgot to follow some of the assembly notes
- I suppose that they did not run  the test procedure using the goldenboard
- On top of this, I need to mentioned that w/ @Fabrizio69 we  found out that there has been a switch of serial number for the 2 2foc board with SN:1561 and SN:1573(goldenboard), so I'd check the others as well
- I'd like to fix the test equipment I've with me thus to try to assemble a second equipment to make easier to support who is gonna make the tests  on the boards for us

Thus, we are just waiting for the reply from `https://mwfep.com/it/` regarding the produced 2foc boards and the assembly notes. In the meanwhile I'd like to have a meeting to discuss regarding what to do with:
- the current test equipment, which should be eventually sent to them for finalizing the tests
- the 2foc boards with the erroneous SN
- the old equipment that can be fixed 

cc: @valegagge @maggia80

MSECode, 2024-10-25:
## Action points

| Problem | Solution | Action point| Link to issue |
|:--------:|:--------:|:--------:|:-------:|
| Procedure instructions to be solved | Remove board revision from instructions and add reference to the Logic (ex. J1) --> WIP | @MSECode and @fgarini need to check this and @MSECode updates the procedure instructions | |
| Phase2 of the procedure does not flash the test fw, thus problem when using the GOLDENBOARD | Add flash of bl and test fw at the  beginning of phase 2| @MSECode will add and @AntonioAzocar will run the whole procedure to check for problems | |
| Missing labelling for `GOLDENBOARD without the TH` | Substitute the GOLDEN without TH with a new one and label this as GOLDENBOARD without THT | @Fabrizio69 will create the entry for the warehouse | |
| Problems will silicon version during phase 1 of the test for some boards | Check if the silicon revision version check fails for newer or older boards --> hw or sw problem?? | Update sw and/or hw  | |
| Considering we wanna have 2 test equipment, one of this 2 is almost complete but we don't know which components are missing | Prepare the BOM for completing the second test equipment so that we can always send one to the tester and keep one in home | @MSECode w/ support of others | |

valegagge, 2024-10-30:
Thanks @MSECode for updating the table with the relative tasks.

We can close this issue in favor of https://github.com/icub-tech-iit/ipts/issues/53 where we are going to address all these tasks.

closed 2024-10-30 by valegagge