mesh-iit/ipts #48: 2FOC test procedure needs to be updated

MSECode, 2024-10-30:
Since there are some of the instruction of the 2FOC test procedure that are outdated we need to update them. This won't block the test but some of the instructions can be misleading to any tester so we need to clarify them well.

fgarini, 2024-11-08:
@MSECode about the PR is there any overlap or duplicate file to manage in the non-legay repo? im referring to https://github.com/icub-tech-iit/electronics-boards/tree/master/2foc/sw

MSECode, 2024-11-12:
Actually yes, but it is just for giving more details to the tester. Basically, when I show in the test procedure how to check which version of the 2foc you are using, I've also add a link to the logic. Therefore, if needed, one can check exactly if the used version of the 2foc is mounting or not the CAN-H/CAN-L resistors. So yes, we should pay attention to changing in the folders of electronics-boards if they might happen in the future.

MSECode, 2024-11-22:
Updates needed are ready. In order to move the related PR to ready I'll wait for the test equipment to arrive from the supplier to test if the modifications done are fine.

MSECode, 2025-04-15:
I've tested the new needles and they are fine. Once the second test equipment has been completed we can close all issues.

closed 2025-06-05 by MSECode