mesh-iit/ipts #49: When using GOLDENBOARD is not possible to flash the test fw on the cores

MSECode, 2024-10-30:
When testing with the GOLDENBOARD only it is not possible to flash on the board the test bootloader and firmware using the test equipment since those steps are not present in the phase2 of the testing procedure. However, since might be quite helpful to re-flash test bl and fw when running the phase2 using the goldenboard we are going to add those steps to the phase2 as well.

cc: @valegagge

MSECode, 2024-11-13:
This PR https://github.com/icub-tech-iit/electronics-legacy/pull/80 brings the updates for solving this issue too.
Basically, I've added the steps for flashing the test fw also at the beginning of `fase2.xml`, which is the only xml that can be completely run if using a goldenboard.

valegagge, 2024-12-20:
I put on hold this issue until the test equipment is available

MSECode, 2025-04-15:
I've updated the `fase2.xml` in order to re-flash the test fw on the board when used with THT components. Moreover, I've updated the .bat files in order to use different class of programmer (PICkit3 and PICkit4) and test it.
I'll run a test on the full `fase2.xml` of the procedure and create the second test equipment to be added to the warehouse.

valegagge, 2025-04-16:
Hi @MSECode , could you please add the reference to your changes? thanks

MSECode, 2025-04-16:
> Hi [@MSECode](https://github.com/MSECode) , could you please add the reference to your changes? thanks

Sure, since now we are keeping that on the sharepoint, I'll add the link.