mesh-iit/ipts #50: Label the `GOLDENBOARD without THT` on the 2FOC test equipment

MSECode, 2024-10-30:
In the test equipment other than the full GOLDENBOARD we have another 2FOC without the THT (through hole) components that can be used a partial goldenboard to check the phase1 of the procedure. Having a board such that in the equipment is useful since it is not possible to run the first phase of the procedure on a board with the THT components mounted. Therefore it is good to have a complete GOLDENBOARD, name as it is, and this partial one. We have currently one in the test equipment but it is quite old and not labelled. We need to bring one from the warehouse, remove the THT components, test that everything works and label it.

cc: @valegagge @Fabrizio69

Fabrizio69, 2024-12-11:
Hi @MSECode , in this moment we haven't available 2FOC boards in the warehouse...
Do you have a board in the Lab? We can remove the THT components in the PROTO Lab

MSECode, 2024-12-11:
Yes, sure I have one. I just need to check which version it is.

Fabrizio69, 2024-12-12:
Thanks @MSECode, when the check is completed, can you give me the board so we can proceed to remove the THT components?

MSECode, 2024-12-12:
The board I have is a revision H, thus it won't be good for doing the second phase of the test procedure. Therefore, it might be good to unmount the THT components. I'll bring it to you.

Fabrizio69, 2024-12-16:
hI @MSECode , I leave the board without THT components on your desk

[image]

MSECode, 2025-06-09:
I've run a test on that board and I noticed that we have only 1 core correctly working. Do we think it is worth to have such a board on the test equipment or the normal golden board is fine?

MSECode, 2025-07-04:
I'd suggest to do not add to the test equipment the golden board without THT. It is enough to have the final golden board.

maggia80, 2025-07-08:
we can ask EES to prepare all the 2FOC fase 1. If they pass the tests we can ask EES to leave one board in fase 1, we will use it as the new goldenboard

Nicogene, 2025-07-09:
Closed since we reached an agreement on this

closed 2025-07-09 by Nicogene