mesh-iit/ipts #51: Check behavior of silicon version check in 2FOC test procedure

MSECode, 2024-10-30:
We need to double check how the check for the silicon version in the phase1 of the test procedure is made on the fw side, so that we can let the test pass when a board has a newer silicon version that the minimum required one. Basically we are not 100% sure if the test step passes for older or newer version. Need to check and eventually update.

cc: @valegagge @fgarini

MSECode, 2024-10-30:
It seems that the silicon version is just match with a specific value:

[image]

but I need to double check this since it is hard to understand which is the source code the test is using. I saw that on another section where we actually have all the test code the silicon version is checked as here:

[image]

thus I'm not sure in which case the test can fail

fgarini, 2024-11-04:
Ciao @MSECode I investigated on this topic
The microcontroller is `DSPIC33FJ128MC802` 
the information about the existing revisions is available in the [silicon errata sheet](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/Errata/80000442k.pdf):
[image]
As you can see the latest revision is the  0x3004. We can think to include in the good version list 0x3005 eventually, even if I'm not expecting new silicon revisions on a quite old uC.

MSECode, 2024-11-04:
Thanks @fgarini, good to know.
So, looking at the version I suppose that we actually do this check:

[image]
 
which matches the version defined in wingst too. I think now to modify the test so that we can accept version not only equal to 0x3003 but also higher. Moreover, we can say that we won't to let boards with a silicon version lower than 0x3003 pass the test if I understood correctly from your notes since having a revision higher than that is critical.

fgarini, 2024-11-04:
Yes I think at the time when the code was written the version 0x3004 wasn't already issued and the reading triggers the error as in the latest snippet you posted.

valegagge, 2024-11-04:
Ok, so we can proceed to add version A5 (0x3004), but I think it is better to avoid adding 0x3005 which currently doesn't exist.

What do you think @MSECode and @fgarini ?

MSECode, 2024-11-06:
I've started to add the updates here: https://github.com/MSECode/electronics-legacy/tree/feature/update2FOCtest
adding the programming of the 2 cores of the 2foc with bootloader and firmware at the beginning of `fase2.xml` and updating unclear instructions in `default.xml`

MSECode, 2024-11-13:
I've studied how to manage this test. Two solutions were possible, one that uses the `operation` type `recvrange` and one that uses the `operation` type `recvpassfail` and adds a method one the fw.
I've decided to use this latter option, which in my opinion is easier to manage and to understand.
Anyway, at this commit https://github.com/MSECode/electronics-legacy/commit/42493588c4c12b7d1b0b204fa32ef4db417ee526 you can find the details about the changes done.

MSECode, 2024-11-22:
After discussion w/ @valegagge, and considering that the 2FOC boards are  soon to be deprecated, we have decided to apply a small workaround to solve the problem of the checking of the silicon version that just need the user to modify the xml file. Therefore, there's no issue with the source code.

MSECode, 2025-06-09:
For solving this problem we have updated the test description.

closed 2025-06-09 by MSECode