mesh-iit/ipts #54: AMC-BLDC - get acquainted with the current status of the test procedure for the board

MSECode, 2025-01-07:
## Task description
Before carrying on with issues: https://github.com/icub-tech-iit/ipts/issues/39 and https://github.com/icub-tech-iit/ipts/issues/38 we need to check the current status of the test procedure for the `amc-bldc` board.

## Definition of Done
Report on the current status of the test procedure for the `amc-bldc` and define which are the following action points.

cc @valegagge

MSECode, 2025-01-13:
After a first run of the test for the AMC_BLD using the latest modification of ipts and the jig for the amc_bldc we have the following result:
- All tests pass but the one of the option bytes as shown by this test report:

[0035_14962.D.txt](https://github.com/user-attachments/files/18397911/0035_14962.D.txt)

Now we need to  check why the test on the option bytes is failing. Moreover we need than to add the new tests.

valegagge, 2025-01-13:
I guess the problem was already noted. see https://github.com/icub-tech-iit/ipts/issues/39.

So now you can move to that issue. what do you think?

MSECode, 2025-01-17:
Yes, that's correct. I've indeed updated that issue. Basically I think that there was some errors in the code that was testing the enabling of the option bytes. Now the test passes with positive outcome. 
I just one to make some other check using the debugger on the source code to be sure that the whole test pipeline is good. 
Once that is done I'll proceed with the rest of the tests.

MSECode, 2025-01-17:
As already commented in #39 all tests are passing fine. I just need to understand couple of stuff related to the code architecture and then this issue can be considered as completed. Once that is done I'll go on with the tests defined in this issue: #53

MSECode, 2025-01-27:
After a discussion with/ @marcoaccame about the architecture and structure of the project of the amc-bldc test firmware we can state the following about the open points that was not totally clear to me:
- most of the embot library functions encapsulates the low level methods correctly, thus for most of the test it is correct to call the higher level embot function, such as when we need the test the HALL sensors 
- some methods are not yet made available by the embot library, e.g. the position of the ABZ encoder, thus we need to leave the test code as it is by calling the HAL methods of the stm32 library 
- the set of the PWM in percentage can be done using the embot library. We only need to set the callback function for saving the values of the currents, which are evaluated by the ADC on a specific loop
- Finally all test have been tested and they pass. We still need to define the threshold for checking the current flowing while feeding the pwm 
- Moreover, we have observed a weird behaviour of the CIN while feeding pwm and we need to test it more in details with the support of @fgarini

MSECode, 2025-01-27:
As said in the previous comment we have updated the code, which is available at this branch: https://github.com/MSECode/icub-firmware/tree/feature/amcbldctest

With this we have completed the firmware part for the tests of the `amc-bldc` board. 
Specifically:
- we have checked that the test for the option bytes works fine
- we have added the tests for the PWM for the 3 phases of the motor by checking also the values of phase currents that flow in the setup loads (we have also checked at the sum of the currents is zero at it should be accordingly to the logic of the circuit). Related to this point we still have an issue with the input current measured by the ADC that we need to investigate more
- on top of this we still need to updated the sw part of the test by adding the new images and instructions for the user that should change the voltage of the power supplier before feeding the PWM in the board 

Once the open points will be closed we can open the PRs and close the following related issues:
- https://github.com/icub-tech-iit/ipts/issues/38
- https://github.com/icub-tech-iit/ipts/issues/39
- https://github.com/icub-tech-iit/dev/issues/762 --> in this issue you can check the observations made by @fgarini  and the plots of the oscilloscope

closed 2025-01-30 by MSECode