mesh-iit/ipts #56: AMC- define the list of tests

valegagge, 2025-02-27:
We need to define the list of tests to implement for the AMC board validation.

**DoD**
test defined

valegagge, 2025-02-27:
HI @MSECode, can yu schedule a meeting with @fgarini before the planning of the next sprint (3->14 March)

thanks!!!

fgarini, 2025-02-28:
@MSECode feel free to reach to me on Teams. 
Just to give an introduction, it will be very similar to what already done for the AMC_BLDC with the addition the SPI tests and the Ethernet connection.

MSECode, 2025-03-03:
After discussion w/ @fgarini and as said in the comment above we will proceed by developing the same tests done for the amc-bldc (the inner functions will be different since the micro are not the same).
Then, once those are done, we are going to develop the following tests:
- ETH (probably in loopback)
- I2C/SPI using the switch on the JIG
- power-supply test and led test on CAN

MSECode, 2025-03-18:
Up to this point I've validated the following tests:
- CAN communication
- LEDs OFF
- LEDs ON
- VIN
- CIN
- Fault OFF
- Fault ON

Here a screenshot that shows the validation of the tests. 
As said [here](https://github.com/icub-tech-iit/ipts/issues/62#issuecomment-2732330340) the tests that are strictly related to the hw need to be  checked completely by using the bsp APIs that communicate directly with the GPIOs. In order to do that we need to use an `amc` project that configures both CM4 and CM7 cores of the board. However, we are currently working on a project where, for simplify  things, enables only the CM7 core. Thus, for the tests that are related to hw checks we call the bsp APIs but we make them reply dummy values. In this way we can check the pipeline of the tests up to the last GPIO request. 

This image shows the validation of the tests:

[image]

MSECode, 2025-03-25:
After the refactoring we still have the implemented tests working:

[image]

MSECode, 2025-06-11:
## June 2025 Update 

Following the development of the amc bsp APIs and the cabling of the AMCx-JIG, we have continued in carrying out the necessary tests for the AMC board.
As shown by the image below, we have introduced some new tests such as:

- FW version test
- VAUX test (which need to be revised with who is responsible for electronic and fw in order to check if it checks what desired)
- ETH Communication
- SPI encoders --> tested all 3 channels by reading from MA730 chip (AEA3 encoder chip)

[image]

Now, we would like to go on with the following tests:
- i2c channels tests --> in this case we need to have an alignment with @marcoaccame  since there are already some bsp APIs but I'd like to add the calls to the amc bsp if possible
- HALL sensor test --> I suppose we have to produce this test on the CM4 test project
- ABZ sensor test --> same as before --> probably we need to develop this test on the CM4 test project
- Option Bytes --> not sure what should I check --> which OB we are interested on ?? --> we just want to check by the test if the flash memory banks are set correctly or we wanna set a check on something else --> for this I need the support from who developed the board project.
- something else we are expected to test @fgarini 

cc: @Nicogene

MSECode, 2025-06-11:
Reference to the following PRs:
- https://github.com/robotology/icub-firmware/pull/612
- https://github.com/icub-tech-iit/ipts/pull/73

valegagge, 2025-06-12:
> SPI encoders --> tested all 3 channels by reading from MA730 chip (AEA3 encoder chip) --> probably for this test I'm thinking to add to the data in output the read positions and pass as input the integer to the type of encoder we wanna test. I think that on the JIG we have both AEA and AEA3 mounted. If so we can test both. What do you think?

yes! Good idea! We need to keep an eye if the same message is used in other boards test in order to be back-compatible

MSECode, 2025-07-25:
# Following Update

Together w/ @marcoaccame we have completed the definition of the bsp for the i2c protocol on the test project of the `amc`.
I've run the whole test suite and it is  fully working.

We still need to finalize some tests which are:

- Test Vin (gives dummy values --> need to be implemented on CM4 core)
- Test Cin (gives dummy values --> need to be implemented on CM4 core)
- Test PWM (need to be coded)

Here the latest test results:

[image]

MSECode, 2025-08-22:
## Update to ipts release v1.6.1

Tests defined up to now are completed and tested:

[image]

We might think to add EEPROM check but we are eventually good to go.

As soon as PR are closed we can close this issue too.

MSECode, 2025-08-25:
Hi @fgarini,
I'll open an issue to just trace that. I've added the EEPROM test to the latest release 1.6.1.
I'll share the link to you as soon as the PR is merged. Up to now the release is in draft. Once the PR in IPTS is merged I'll undraft the release. 
Thus, we have the eeprom check as well. 
Are we good, aren't we? Or do we wanna plan other tests for further releases ?

closed 2025-08-26 by MSECode