mesh-iit/ipts #71: Fabrizio will try the 2foc testing procedure

Nicogene, 2025-06-05:
### Task description

Once @MSECode finished tweaking the 2foc procedure:
- https://github.com/icub-tech-iit/ipts/issues/70

@Fabrizio69 will try it by his own, and he will give feedbacks for possible improvements

### Definition of done

@Fabrizio69 did the testing procedure of 2foc

Nicogene, 2025-07-02:
Actually it is @filippoborgogni that is trying the procedure:
- https://github.com/icub-tech-iit/proto/issues/3343

filippoborgogni, 2025-07-02:
I tried the procedure and I just posted a comment where I report what needs to be changed to make everything clearer and faster; when everything is changed I will try out the test equipment one last time before closing the issue!

cc @Fabrizio69 @Nicogene @MSECode